ros-drivers/nmea_navsat_driver #13: Driver does not support GGA quality values > 5

ablakey, 2015-06-07:
The use of GGA quality values greater than 5 is becoming more common. In particular, Novatel boards announce WAAS as a value of 9.  In this case, the driver would report a -1 for current_fix.status.status. Our solution was to reproogram the GPS to report all quality values of 9 as 2 instead.  

I believe this merits discussion on the if/how the driver could be changed to be more accommodating.

My concerns are:
1. The GGA quality values are being abused more and more and the mappings between quality and NavSatStatus.status might not be static.

2. The NavSatStatus.status message is ill-equipped for the growing variety of solution options.  Currently we can only declare NO_FIX, FIX, GBAS_FIX, SBAS_FIX, which leaves the nmea_navsat_driver little way to communicate the solution type.

mikepurvis, 2015-10-16:
Maybe there needs to be an option for the user to submit a rosparam of the mapping table that's appropriate to their device?

Alternatively, use the nmea_sentence topic with an external node to extract the quality (and any other vendor-specific data you care about), and publish it in a form suitable to your application?

ablakey, 2015-10-17:
I think an optional remapping via rosparam is a good option because of the lack of a formal or even de-facto standard. I wouldn't want to impose specific meanings on developers.  This feature would have allowed us to simply remap the values rather than fork or reprogram the NovAtel.

This also allows us to not worry about future mappings so much, so when GNSS vendors begin making use of additional quality integers, the hard coded mapping doesn't restrict use.

The alternative solution you present seems like a viable resolution to me. Though I think I'm more concerned about what/if anything needs to be amended in this driver to keep it usable into the future.

PaulBouchier, 2016-03-20:
@ablakey can you please propose a parameter that would remap quality in the way you want. Also, if Novatel is using 9 for WAAS it seems we should add 9 as a supported quality and give it a parameterized default EPE, as I'm doing with the other quality values per issue 17. Do you agree?

Can you test any changes with a Novatel receiver?

ablakey, 2016-03-20:
I think the simplest implementation would be a map of arrays. Any fix enum in the map has their defaults overridden. It's one parameter and explicit, which is good. Nested collections is not the best but I see no other way around it.

For example, I might want to override two of them with a few more integers:
```
---
  FIX_STATUS_MAP: 
    STATUS_SBAS_FIX: 
      - 2
      - 6
    STATUS_GBAS_FIX: 
      - 4
      - 5
      - 8
      - 9
```

I don't know if we still have a WAAS enabled GNSS but I can certainly test the business logic with a NovAtel unit.  Happy to author this modification too!

PaulBouchier, 2016-03-27:
Andrew - I'd appreciate you providing the modification to support remapping GPS quality values. You have an idea in mind that I don't quite get. It would be best to either fork my fork, which has support for GPS quality 9 and the new parameters that support correct EPE reporting, or wait until Eric has pulled my work into the top of tree. At a minimum, please review what I did so we don't stamp on each other. My fork is at https://github.com/PaulBouchier/nmea_navsat_driver

ablakey, 2016-03-27:
Hi Paul:

Good idea.  I'll likely approach this work one evening this week. It sounds like forking, or at least collaborating on your fork makes the most sense.

From a quick peek at your work, I'm thinking we might want to consider the value of @mikepurvis's suggestion of allowing a custom override of NMEA mappings (and epe quality?) rather than explicitly defining the mapping in the code. While there might be some common conventions or pseudo-standards for the integer mappings, it may still fall short in cases where the mappings vary from device to device.

A bonus effect of this approach is that we do not alter the behaviour of the driver, possibly breaking fragile implementations that depend on the current mappings.

PaulBouchier, 2016-03-27:
Thanks for working on this.

Oops - I need to merge my changes into indigo, currently they're only on jade.

I thought the idea was to map one gps_quality to another, with the intent that the second would have an associated epe that would be published in covariance, and to do this remapping with parameters. Is that right?

I thought the use of this would be for some receiver that may emit gps_quality = 3 (an unsupported value), and one could remap it to, say, 2, which has an associated meaning, status, and default epe. Is that right?

About the only thing gps_quality is used for is to set the default epe that feeds into covariance, and to set the gps fix status. My implementation of support for covariance allows those default epe's to be changed with parameters.  I don't understand "epe quality" when you say "allowing a custom override of NMEA mappings (and epe quality?) rather than explicitly defining the mapping in the code". Did you mean gps_quality? epe default can receive a custom override via parameters as now implemented.

Just trying to make sure we're on the same page. As you say, keeping the default behavior of the driver has a lot of merit.

PaulBouchier, 2016-03-28:
Added support for gps_quality 9 in pull request #19 for jade & #20 for indigo

ablakey, 2016-03-28:
Yeah I think we're on the same page, just framing the problem from different perspectives.  

The bottom line is that with the current proposed changes, if your quality is any value that is NOT:  1,2,4,5,9  then it sets:  `current_fix.status.status = NavSatStatus.STATUS_NO_FIX`   which is wrong.

I was thinking about a way to have default mappings for 1,2,4,5,9  but allow a user to provide params that override these.  Pseudocode:

```
custom_qual_map = rospy.get_params('override_qualities')
qual_values = custom_qual_map.get(gps_qual) or default_qual_map.get(gps_qual)
if qual_values:
    self.fix_type = qual_values['fix_type']
    self.default_epe = qual_values['default_epe']
    current_fix.status.status = qual_values['navsat_status]
else:
    self.fix_type = 'Unknown'
    current_fix.status.status = NavSatStatus.STATUS_NO_FIX
```

Obviously less awful var names but you probably get the gist.

    

We can definitely consider it for this or a different pull request.

PaulBouchier, 2016-03-29:
OK now I think I get the idea better. Makes sense. FYI I submitted a pull request for the work I did to try & at least get it into the source repo to support the Skytraq RTK GPS receiver which they're announcing this week and stating ROS support.

ablakey, 2016-04-05:
@PaulBouchier I've got some time this week to approach this.  What do you think the best method is for moving forward? Should I fork your fork and raise a PR for you to fold into the existing PRs?  Or should I await your PRs to be merged first? They'll conflict considerably, so I suspect doing it in parallel is a bad idea.

PaulBouchier, 2016-04-06:
@mikepurvis @ericperko Can you guys please advise on the timeframe for merging pull requests 19 & 20. Do you have any concerns with it? Andrew needs to know what to base his further work on, and it should be made to properly support the new RTK GPS modules.

mikepurvis, 2016-04-06:
I've reviewed #19. Once we get that into a state we like, it should be cherry-picked back to Indigo (rather than merged).

@ablakey Suggest you work from Paul's branch with an expectation that you'll rebase your work to ros-drivers and merge there when complete.

evenator, 2019-05-14:
I'm closing this issue as I believe it was mostly solved in PR #46. If there are specific issues that remain unaddressed, please file new issues.

closed 2019-05-14 by evenator