ros-drivers/nmea_navsat_driver #15: GNSS strings

emilihb, 2015-06-10:
The current parser assumes $GPxxx strings only, which doesn't work with ublox 8. May I suggest to change the regular expression to '^\$G[N|P].*\*[0-9A-Fa-f]{2}$' to fix this?

ablakey, 2015-10-17:
Do you have example strings emitted by the ublox 8? I'm curious how they're departing from the NMEA format.  If I can read your regex properly,  they're often emitting $GNxxx  ?

valschmidt, 2016-06-29:
It is also not uncommon for GPS aided INSs to produce an IN prefix (e.g. $INGGA). So the expression could be modified to  '^\$[I|G][N|P].\[0-9A-Fa-f]{2}$' (untested).

johschmitz, 2016-07-14:
GP is for GPS only GN is for multi constelation GNSS systems (GPS,GLONASS,GALILEO)
See the NMEA specification.

Anybody has a regex which can also parse ublox binary messages?

blmhemu, 2017-01-08:
Go to /opt/ros/indigo/lib/python2.7/dist-packages/libnmea_navsat_driver (may be different for yours just type 'locate parser.py' ) and replace the parser.py in that folder with the one present in this git "https://github.com/ros-drivers/nmea_navsat_driver/blob/jade-devel/src/libnmea_navsat_driver/parser.py").

jack-oquin, 2017-01-09:
Note that modifying files in the ``/opt/ros/...`` directory tree is only a temporary expedient. Future package updates can overwrite them.

evenator, 2018-08-11:
This appears to have been fixed in 904ea247aa0c5305facf71811defc846394b28d3

closed 2018-08-11 by evenator