ros-drivers/nmea_navsat_driver #24: frame_id pre-appended by a forward slash

shoemakerlevy9, 2016-10-06:
I have been observing that the nmea_serial_driver node pre-appends a forwardslash onto whatever frame_id is specified, even the default of "gps" becomes "/gps". This causes errors in nodes that try to use this transform. I am specifically using robot_lacalization's gps_transform_node which cannot support frame_id's starting with a forwardslash.

roccogalati, 2017-03-21:
I'm having the same problem! I posted it as an issue, I hope to find a solution since I'm stuck with my research work!

samkys, 2018-08-10:
Any news/changes yet? What is the preferred method to fix this as I need this to work as well and just have been using a "dirty hack" by assuming the user inputs the frame_id they desire and thus comment out these sections of code inside  nmea_navsat_driver/src/libnmea_navsat_driver/driver.py :

```    def get_frame_id():
        frame_id = rospy.get_param('~frame_id', 'gps')
        if frame_id[0] != "/":
            """Add the TF prefix"""
            prefix = ""
            prefix_param = rospy.search_param('tf_prefix')
            if prefix_param:
                prefix = rospy.get_param(prefix_param)
                if prefix[0] != "/":
                    prefix = "/%s" % prefix
            return "%s/%s" % (prefix, frame_id)
        else:
return frame_id```

Changed to:
```    def get_frame_id():
        frame_id = rospy.get_param('~frame_id', 'gps')
#        if frame_id[0] != "/":
#            """Add the TF prefix"""
#           prefix = ""
#            prefix_param = rospy.search_param('tf_prefix')
#           if prefix_param:
#                prefix = rospy.get_param(prefix_param)
#               if prefix[0] != "/":
#                    prefix = "/%s" % prefix
#           return "%s/%s" % (prefix, frame_id)
#        else:
return frame_id```

evenator, 2018-08-11:
This will be fixed shortly by #33. See the comments on that PR for more details.

evenator, 2018-10-14:
Closed as fixed by #33.

closed 2018-10-14 by evenator