ros-drivers/nmea_navsat_driver #29: Altitude vs Ellipsoid Height

valschmidt, 2017-02-06:
It seems to me the nmea_navsat_driver incorrectly calculates makes a correction for the `altitude` portion of the NavSatFix message. Specifically, here's the code from "diver.py"

```
# Altitude is above ellipsoid, so adjust for mean-sea-level
        altitude = data['altitude'] + data['mean_sea_level']
        current_fix.altitude = altitude
```

The NavSatFix message definition is the following:
```
# Altitude [m]. Positive is above the WGS 84 ellipsoid
# (quiet NaN if no altitude is available).
float64 altitude
```

The GGA sentence "ellipsoidal height" produces exactly that, height relative to the WGS 84 ellipsoid, positive up. The choice of "altitude" here to describe this quantity is probably mis-leading, but in any case, based on the NavSatFix definition, one should report  the ellipsoidal height field with no correction, imo. 

[I will admit, for many applications, altitude above mean sea level, as it is calculated here, is probably more intuitive. But for anyone doing any kind of precision navigation, it could be very misleading.]

jack-oquin, 2017-02-06:
I agree with your interpretation of the NavSatFix message.

evenator, 2018-08-14:
Fixed by #36

closed 2018-08-14 by evenator