ros-drivers/nmea_navsat_driver #31: frame_id include a "/" which cannot work in tf2

roccogalati, 2017-03-21:
Hello guys,

I'm having problems with the `nmea_navsat_driver` under Indigo since it publishes a frame_id as /gps which includes a backslash which is not allowed in `tf2` standard.
I tried to launch the driver with the " `_frame_id:=gps`" parameter but it still continues to add the backslash to the frame name. 
Is it possible to change this, please? I cannot correctly run my ROS codes since I always get this error:
` Could not obtain transform from /gps to base_link. Error was Invalid argument "/gps" passed to lookupTransform argument source_frame in tf2 frame_ids cannot start with a '/' like: `

Thank you!

jack-oquin, 2017-03-23:
That does look like a bug.

If you submit a pull request with a fix for it, you're likely to get it done sooner. (I am not the maintainer of this package.)

evenator, 2017-12-15:
This is a duplicate of #24

closed 2017-12-15 by evenator