ros-drivers/nmea_navsat_driver #35: tf has been "deprecated" in favor of tf2

Pietro-Colombo, 2017-04-27:
Since ROS Hydro, tf has been "deprecated" in favor of tf2.
http://wiki.ros.org/tf2/Migration

but nmea_navsat_driver use tf.

https://github.com/ros-drivers/nmea_navsat_driver/blob/jade-devel/src/libnmea_navsat_driver/driver.py

at line 177:

@staticmethod
    def get_frame_id():
        frame_id = rospy.get_param('~frame_id', 'gps')
        if frame_id[0] != "/":
            """Add the TF prefix""" 
            prefix = "" 
            prefix_param = rospy.search_param('tf_prefix')
            if prefix_param:
                prefix = rospy.get_param(prefix_param)
                if prefix[0] != "/":
                    prefix = "/%s" % prefix
            return "%s/%s" % (prefix, frame_id)
        else:
            return frame_id

because i have problem to use ekf_filter with tf.

Thanks

jack-oquin, 2017-04-27:
Lots of packages still use ``tf``.

Specifically, what is not working for you here?

Pietro-Colombo, 2017-04-27:
i see other issues and other person have my same problem and they suggest this:
https://github.com/ros-drivers/nmea_navsat_driver/pull/33

Because in my case I want implement ekf-filter with **robot_localisation Package**
this package want frames without a slash.

http://answers.ros.org/question/259756/navsat_transform_node-tf-has-two-or-more-unconnected-trees/

griswaldbrooks, 2017-04-27:
I also had the same issue, because I was trying to use `robot_localization`

jack-oquin, 2017-04-27:
OK, I think several issues are being confused here.

1. Frame ID strings should no longer contain a leading '/'
2. The old ``tf`` interface is still supported
3. Some systems still need ``tf_prefix``, and no replacement is available

evenator, 2017-12-15:
Related: #24

evenator, 2018-08-11:
I agree with @jack-oquin's assessment.

1. The leading slash issue will be fixed shortly by #33.
2. The old `tf` package still exists and is still supported. If someone wants to take on the task of "modernizing" this repo to use `tf2`, I would be happy to review that PR 🙂 
3. The solution in #33 removes the requirement for a leading slash without removing the `tf_prefix` that some systems need.

closed 2018-08-11 by evenator