ros-drivers/nmea_navsat_driver #39: Frame id must start with "/".

blmhemu, 2017-12-10:
This is changed in the latest tf2 release. It gives warning when a slash is present.

evenator, 2017-12-15:
There was some discussion about this previously in #35, but it sounds like there's now a deprecation warning. I'm hesitant to change the behavior in existing releases for the reason that @jack-oquin stated in that discussion--I suspect some users are expecting the leading slash. I'd like to hear what others think about this. I think the best course of action is probably to make this change in lunar only.

evenator, 2017-12-15:
See also #40

evenator, 2017-12-15:
After looking through the issues and PRs here, I'm closing this as a duplicate of #24. Please continue all discussion there.

closed 2017-12-15 by evenator