ros-drivers/nmea_navsat_driver #41: nmea_topic_driver is not publishing anything

taihup, 2018-01-24:
Hi!

I created a rosbag that contains NMEA messages ([nmea_msgs/Sentence](http://docs.ros.org/api/nmea_msgs/html/msg/Sentence.html)), and I would like to generate [sensor_msgs/NavSatFix](http://docs.ros.org/api/sensor_msgs/html/msg/NavSatFix.html) messages.

I am working on simulating time, so I run:

    rosparam set use_sim_time true

Then, I play the rosbag thats contains the NMEA messages

    rosbag play --clock nmea_gps.bag

Then, I start the nmea_topic_driver to generate publish [sensor_msgs/NavSatFix](http://docs.ros.org/api/sensor_msgs/html/msg/NavSatFix.html) on /fix topic:

    rosrun nmea_navsat_driver nmea_topic_driver

Finally, when I check with rostopic if there are message on /fix topic

    rostopic echo /fix

    WARNING: no messages received and simulated time is active.
    Is /clock being published?

Any, idea of what is going on?

I check if nmea_topic_driver was subscribed to the NMEA messages properly, and it looks fine:

    rostopic info /nmea_sentence 
    Type: nmea_msgs/Sentence
    
    Publishers: 
     * /play_1516740862743383627 (http://MS-7A15:46035/)
    
    Subscribers: 
     * /nmea_topic_driver (http://MS-7A15:46775/)

I also checked if the NMEA messages were right, and they look fine too:

    rostopic echo /nmea_sentence 
    
    header: 
      seq: 0
      stamp: 
        secs: 1514301945
        nsecs: 488000000
      frame_id: "GPS-RTK"
    sentence: "$GPRMC,152545.80,A,3301.9913392,S,06053.0775644,W,0.12,0.00,261217,0.0,E,D*3F\r\n"
    ---
    header: 
      seq: 1
      stamp: 
        secs: 1514301945
        nsecs: 488000000
      frame_id: "GPS-RTK"
    sentence: "$GPGGA,152545.80,3301.9913392,S,06053.0775644,W,5,09,1.0,52.356,M,17.442,M,0.2,*72\r\
      \n"
    ---
    header: 
      seq: 2
      stamp: 
        secs: 1514301945
        nsecs: 488000000
      frame_id: "GPS-RTK"
    sentence: "$GPGSA,A,3,02,12,24,25,29,,,,,,,,12.0,1.7,11.8,1*29\r\n"
    ---
    header: 
      seq: 3
      stamp: 
        secs: 1514301945
        nsecs: 488000000
      frame_id: "GPS-RTK"
    sentence: "$GLGSA,A,3,75,76,86,87,,,,,,,,,43.1,15.6,40.1,2*03\r\n"
    ---
    header: 
      seq: 4
      stamp: 
        secs: 1514301945
        nsecs: 488000000
      frame_id: "GPS-RTK"
    sentence: "$GPGSV,2,1,05,02,52,107,35,12,61,145,38,24,49,007,37,25,48,217,36,1*6A\r\n"
    ---
    header: 
      seq: 5
      stamp: 
        secs: 1514301945
        nsecs: 488000000
      frame_id: "GPS-RTK"
    sentence: "$GPGSV,2,2,05,29,47,283,41,,,,,,,,,,,,,1*55\r\n"
    ---
    header: 
      seq: 6
      stamp: 
        secs: 1514301945
        nsecs: 488000000
      frame_id: "GPS-RTK"
    sentence: "$GLGSV,1,1,04,75,42,053,37,76,75,160,36,86,72,178,39,87,41,315,35,1*7F\r\n"
    ---
    header: 
      seq: 7
      stamp: 
        secs: 1514301945
        nsecs: 488000000
      frame_id: "GPS-RTK"
    sentence: "$GPGST,152545.80,0.090,,,,0.081,0.061,0.194*78\r\n"
    ---
    header: 
      seq: 8
      stamp: 
        secs: 1514301945
        nsecs: 488000000
      frame_id: "GPS-RTK"
    sentence: "$GPVTG,0.00,T,0.00,M,0.12,N,0.06,K,A*26\r\n"
    ---

The output of `rosnode info /nmea_topic_driver` is:

    rosnode info /nmea_topic_driver
    
    Node [/nmea_topic_driver]
    Publications: 
     * /fix [sensor_msgs/NavSatFix]
     * /rosout [rosgraph_msgs/Log]
     * /time_reference [sensor_msgs/TimeReference]
     * /vel [geometry_msgs/TwistStamped]
    
    Subscriptions: 
     * /clock [rosgraph_msgs/Clock]
     * /nmea_sentence [nmea_msgs/Sentence]
    
    Services: 
     * /nmea_topic_driver/get_loggers
     * /nmea_topic_driver/set_logger_level
    
    
    contacting node http://MS-7A15:34339/ ...
    Pid: 28973
    Connections:
     * topic: /rosout
        * to: /rosout
        * direction: outbound
        * transport: TCPROS
     * topic: /nmea_sentence
        * to: /play_1516817666576416404 (http://MS-7A15:38155/)
        * direction: inbound
        * transport: TCPROS
     * topic: /clock
        * to: /play_1516817666576416404 (http://MS-7A15:38155/)
        * direction: inbound
        * transport: TCPROS

Thanks in advance

evenator, 2018-02-17:
Can you share your bag so I can try to replicate the problem?

taihup, 2018-02-17:
Thanks for the answer. The issue was solved when we removed the CR (Carriage Return) and the LF (Line Feed) control characters from the nmea sentences.

closed 2018-02-17 by taihup

orz-000, 2019-03-10:
I have same problem. What are CR and LF?

taihup, 2019-03-11:
You have to delete them from the nmea sentences. These characters are invisible, so you have to remove them from the nmea sentences. They are located at the end of each nmea sentence.

robograffitti, 2022-11-03:
I encountered the same problem and resolved with the same way.

However, I'm wondering why nmea_navsat_driver does not accept a sentence with CR and LF.

The format guidline let us put them in the end:
https://en.wikipedia.org/wiki/NMEA_0183

josephduchesne, 2025-01-18:
I ran into this today as well. Easy to fix, but a pain. It would be nice if the node stripped them off the end.