ros-drivers/nmea_navsat_driver #43: I meet a error : invalid checksum

hoppss, 2018-03-27:
" Received a sentence with an invalid checksum"

I found from roswiki that someone also meet this error, but unsolved.

Does anyone know how to solve it ?
Thank you.

hoppss, 2018-03-30:
Thanks for a TsingHua university doctor , whos first name is GUANG.
This error is solved by redefine /opt/ros/indigo/lib/python2.7/dist.../libnmea_navstat_driver/parse.py
the function of "parse_nmea_sentence" 's regex part is fault. 
"def parse_nmea_sentence(nmea_sentence):
    # ( original code --- is invalid )Check for a valid nmea sentence
    #if not re.match('^\$GP.*\*[0-9A-Fa-f]{2}$', nmea_sentence):
    #    logger.debug("Regex didn't match, sentence not valid NMEA? Sentence was: %s"
    #                 % repr(nmea_sentence))
    #    return False
    
    #-----ke yong ----2018.3.30
    if nmea_sentence.find('GGA')<0:
        logger.debug("Regex didn't match, sentence not valid NMEA? Sentence was: %s"
                     % repr(nmea_sentence))
        return False
    else:
        print("GGA PAST")

    fields = [field.strip(',') for field in nmea_sentence.split(',')]
"

evenator, 2018-03-31:
Can you post an example of a NMEA sentence that fails to parse with the previous parser but passes with the changes?

droter, 2018-05-12:
Here are some messages that fail to parse:

[WARN] [1526166711.226185]: Received a sentence with an invalid checksum. Sentence was: '\x00$GPGGA,231152.200,4045.8557269,N,08706.9263476,W,1,04,4.0,204.01,M,,M,,*66'
[WARN] [1526166711.337624]: Received a sentence with an invalid checksum. Sentence was: '\x00$GPGGA,231152.300,4045.8564851,N,08706.9271579,W,1,04,4.0,203.45,M,,M,,*6C'
[WARN] [1526166711.433838]: Received a sentence with an invalid checksum. Sentence was: '\x00$GPGGA,231152.400,4045.8573472,N,08706.9282339,W,1,04,4.0,203.19,M,,M,,*67'
[WARN] [1526166711.530014]: Received a sentence with an invalid checksum. Sentence was: '\x00$GPGGA,231152.500,4045.8573405,N,08706.9290835,W,1,04,4.0,205.47,M,,M,,*6F'

oxoocoffee, 2019-03-30:
I noticed as well. Were you able to find out what the problem it. I quicly looked at your code and I do not see anything problematic at first. Open serial driver, dealline, and check checksum. Here are few lines from name serial driver log

`Received a sentence with an invalid checksum. Sentence was: '$GPGGA-34.5,M,4.3,0138*77'`
`Received a sentence with an invalid checksum. Sentence was: '$GPGGA,1322350.9,1.3,1*28'`
`Received a sentence with an invalid checksum. Sentence was: '$GPGGA,1323'`
`Received a sentence with an invalid checksum. Sentence was: '$GPZDA,132339.00,*79'`
`Received a sentence with an invalid checksum. Sentence was: '$GPGGA,132410.,03,10,12,14,18,22,25,26,31,,,1.4,0.8,1.2,1*20'`
`Received a sentence with an invalid checksum. Sentence was: '$GPGGA,132442.40,4303.8003354,N,08753.5276990,W,2,09,0.9,193.88*77'`
`Received a sentence with an invalid checksum. Sentence was: '$GPGGA,132514.10,4303.8003214,N,08753.52768,M,-34.5,M,7.6,0138*70'`
`Received a sentence with an invalid checksum. Sentence was: '$GPGGA,132545.90,4303.8003381,N,08753.5276579,W,2,09,0.9,193.8,M,-34.5,M,5.9,0138*7593.8,M,-34.5,M,4.3,0138*76'`
`Received a sentence with an invalid checksum. Sentence was: '$GPGGA,132617.60,4303.8003512,N,08753.5276330,W,2,09,0.9,193.8,M,-34.5,M,7.6,014,22,25,26,31,,,,1.5,0.9,1.2,1*29'`
`Received a sentence with an invalid checksum. Sentence was: '$GPGGA,132649.40,4303.8003403,N,08753.5276156,W,2,10,0.8,194.0,M,-34.5,M,34.5,M,6.9,0138*77'`
`Received a sentence with an invalid checksum. Sentence was: '$GPGGA,132752.70,4303.8003248,N,08753.5276245,W,2,09,0.9,194.0,M,-34.5,M,6.7,0138*794.0,M,-34.5,M,5.1,0138*79'`
`Received a sentence with an invalid checksum. Sentence was: '7.0,0138*7A'`
`Received a sentence with an invalid checksum. Sentence was: '$138*71'`

oxoocoffee, 2019-04-16:
I have looked at this and try to understand it. After some testing and research here is what I was able to get.

Looks like it is consumer/producer problem with some units. Some vendors actually acknowledge this in their documentation. Here is one Hemisphere model A101 and A222 (User Guide Revision: A2 May 7, 2018) Page 39 section "Random binary data from A222" stats:

`• Verify the baud rate settings match`
`• Potentially, the volume of data requested to be output could be higher than the current baud rate supports. Try either using a higher baud rate for communications or decreasing the number of messages and/or baud rates`

So with my tastes and running such tests overnights I was able to get consistent problems reported here. If the baud rate is too slow for message rate messages from serial do come corrupt therefore CRC check fails.
Some other units I tested have the same issue but do not mention anything like that in their documentation. One unit I tested did not have that problem. I could set baud rate 4800 or 9600 and messages were coming with proper CRC. This would point to "problematic" firmware implementation of circular buffer or whatever buffering technique is used. Increasing baud rate is one part of the problem. If your CPU is overvalued for a long time the same problem will accrue with such units. Messages are not removed fast enough from UART. I was able to reproduce this as well with high baud rate and VERY high host CPU usage. But in this case, it does not happen that often.

Tested on: 
`Ubuntu 18.04`
`ROS(melodic)`
`pyserial(3.4)`
`nmea_navsat_driver(master cloned 2 week ago)`

Baud Rates:
`4800 - 19200`

Message rates: 
`GPGGA: 1 - 20 Hz`
`GPRMC: 1 - 20 Hz`
`GPGST:  1Hz`

droter, 2020-03-24:
Example of bad checksum with binary data before nmea message:

L����*$GNRMC,150541.20,A,4832.56003,N,11753.56027,W,0.058,,240320,,,F,V*07
$GNRMC,150541.20,A,4832.56003,N,11753.56027,W,0.058,,240320,,,F,V*07
$GNGGA,150541.20,4832.56003,N,11753.56027,W,5,12,0.58,525.0,M,-17.0,M,0.2,0000*53
$GNGGA,150541.20,4832.56003,N,11753.56027,W,5,12,0.58,525.0,M,-17.0,M,0.2,0000*53
L����*$GNRMC,150541.30,A,4832.56003,N,11753.56027,W,0.037,,240320,,,F,V*0F
$GNRMC,150541.30,A,4832.56003,N,11753.56027,W,0.037,,240320,,,F,V*0F
$GNGGA,150541.30,4832.56003,N,11753.56027,W,5,12,0.58,525.0,M,-17.0,M,0.3,0000*53
$GNGGA,150541.30,4832.56003,N,11753.56027,W,5,12,0.58,525.0,M,-17.0,M,0.3,0000*53
L����*$GNRMC,150541.40,A,4832.56003,N,11753.56027,W,0.039,,240320,,,F,V*06

evenator, 2020-10-11:
This would probably be fixed by #55, but I think the author of that PR has abandoned it.

halqupati, 2021-08-15:
how you are connect your GPS with laptop?
This error means your devices are not in same band rate.

And the best solution is to use Arduino with GPS sensor.

closed 2022-02-06 by evenator