ros-drivers/nmea_navsat_driver #45: Support for GPS ublox 7n

dahlan77, 2018-04-15:
with nmea navsat driver why when i launch,,,cannot show the latitude and longitude when i use ublox 7n for this package

what should i do to reedit this package so i can use this device correctly

evenator, 2018-04-15:
Unfortunately, I don't have a u-blox 7N to test with. Is your device configured to output NMEA sentences? If so, please capture a log of NMEA sentences, i.e. by using `cat` to pipe the serial port to a text file.

evenator, 2018-05-06:
It looks like the driver is only getting partial NMEA sentences for some reason. That appears to be the tail end of a sentence.

Can you please capture a log of the raw output from the serial port, i.e. by using `cat` to pipe the serial port to a text file? For example, if your device's serial port address was `/dev/ttyUSB0`, you would start the driver, then, in another terminal, run this command:

```bash
cat /dev/ttyUSB0 > gps_terminal_log.txt
```
After a few seconds, hit `CTRL+C` to stop logging, and attach the text file `gps_terminal_log.txt` to this issue so I can take a look at it.

zlacelle, 2018-09-10:
I might have the same issue, so here's some data:

`[WARN] [1536615680.992768]: Received a sentence with an invalid checksum. Sentence was: '\xb5b\x01\x064\x008\x0e\xce\t\x05\xe6\x01\x00\xe2\x07\x02\xdd\xf58\x8f\x06\x13z\x1e\xe3\xe4\xa2\xc1\x17\xb3\x00\x00\x00\x02\x00\x00\x00\x03\x00\x00\x00\x03\x00\x00\x001\x00\x00\x00<\x01\x02\x03n1\x02\x00\x89T\xb5b\x01\x04\x12\x008\x0e\xce\tK\x01<\x01c\x00d\x00,\x01#\x01G\x00\x1cA\xb5b\x01 \x10\x008\x0e\xce\t\x05\xe6\x01\x00\xe2\x07\x12\x07\x03\x00\x00\x00?\xa3$GPRMC,214121.00,A,3855.33603,N,07712.24087,W,0.096,,100918,,,A*6D'
[WARN] [1536615681.999567]: Received a sentence with an invalid checksum. Sentence was: "\xb5b\x01\x064\x00 \x12\xce\t{\xf0\x01\x00\xe2\x07\x03\xdd\xf38\x8f\x06\x0fz\x1e\xe3\xe5\xa2\xc1\x17\xba\x00\x00\x00\x00\x00\x00\x00\xfe\xff\xff\xff\xff\xff\xff\xff\x13\x00\x00\x00\x13\x01\x02\x06n1\x02\x00\xa3\x8f\xb5b\x01\x04\x12\x00 \x12\xce\t'\x01\x13\x01i\x00\x02\x01a\x00J\x00?\x00\xb2\xe6\xb5b\x01 \x10\x00 \x12\xce\t{\xf0\x01\x00\xe2\x07\x12\x07\x03\x00\x00\x00\xabU$GPRMC,214122.00,A,3855.33602,N,07712.24089,W,0.044,,100918,,,A*6E"
[WARN] [1536615683.001205]: Received a sentence with an invalid checksum. Sentence was: '\xb5b\x01\x064\x00\x08\x16\xce\t\xf3\xfa\x01\x00\xe2\x07\x03\xdd\xf48\x8f\x06\x0bz\x1e\xe3\xe9\xa2\xc1\x17\xb9\x00\x00\x00\x00\x00\x00\x00\xfc\xff\xff\xff\x04\x00\x00\x00\x15\x00\x00\x00\x91\x00\x02'
[WARN] [1536615683.005779]: Received a sentence with an invalid checksum. Sentence was: 'n1\x02\x00\x9a4\xb5b\x01\x04\x12\x00\x08\x16\xce\t\xa2\x00\x91\x00H\x00x\x00Q\x00<\x005\x00\xc1\xb7\xb5b\x01 \x10\x00\x08\x16\xce\t\xf3\xfa\x01\x00\xe2\x07\x12\x07\x03\x00\x00\x00\x19\x1f$GPRMC,214123.00,A,3855.33602,N,07712.24088,W,0.019,,100918,,,A*66'
[WARN] [1536615684.009866]: Received a sentence with an invalid checksum. Sentence was: '\xb5b\x01\x064\x00\xf0\x19\xce\tk\x05\x02\x00\xe2\x07\x03\xdd\xf68\x8f\x06'
[WARN] [1536615684.010784]: Received a sentence with an invalid checksum. Sentence was: 'z\x1e\xe3\xeb\xa2\xc1\x17\xb8\x00\x00\x00\x01\x00\x00\x00\xfd\xff\xff\xff\x02\x00\x00\x00\x18\x00\x00\x00\x91\x00\x02'`

I'm attaching some raw data as well.

[ublox.minicom.cap.txt](https://github.com/ros-drivers/nmea_navsat_driver/files/2368503/ublox.minicom.cap.txt)

evenator, 2018-09-29:
@zlacelle, looking at your captured data, I see a mixture of ASCII messages (GPVTG, GPGGA, GPGSA, GPGSV, GPGLL) and what appears to be a binary message. I'm not sure what the binary message, but I expect that's the source of the error messages. Is there a setting on your receiver to disable those messages?

kmikaz51, 2018-11-29:
Hi,

I am using this package on ROS Melodic with a ublox neo M8N and I got the same issue.
After checking the frames on the serial, I saw binary inside.

But I tried the version of the driver on the branch Jade-devel wich works and gives me a fix.

After looking at the code, I think the Issue comes from this line:
https://github.com/ros-drivers/nmea_navsat_driver/blob/a3a3d10f7573ced06b13e879061896020e9adb12/src/libnmea_navsat_driver/parser.py#L143 wich is this line on Jade-devel: https://github.com/ros-drivers/nmea_navsat_driver/blob/24a9cb7a03eb0162075872842678996d118e71d1/src/libnmea_navsat_driver/parser.py#L133

Can we merge this line in master or create a melodic branch ?

schadocalex, 2018-11-29:
I don't have a GPS with me to test but here's my notes:

@zlacelle I did try to put your files into the driver, with a code like:
```python
driver = RosNMEADriver()
f = open("ublox.minicom.cap.txt","r")
if f.mode == "r":
    lines = f.readlines()
    for line in lines:
        driver.add_sentence(line.strip(), "gps")
f.close()
```

and it does work. The last topic published is:

```text
header: 
  seq: 0
  stamp: 
    secs: 1543517159
    nsecs: 789355993
  frame_id: "gps"
status: 
  status: 1
  service: 1
latitude: 38.9222715
longitude: -77.2040181667
altitude: 86.8
position_covariance: [0.008100000000000001, 0.0, 0.0, 0.0, 0.008100000000000001, 0.0, 0.0, 0.0, 0.12960000000000002]
position_covariance_type: 1
```

The warn about bad checksum is real but I think this is not _only_ binary content that provokes the error. In your file there are valid lines.

I don't see any difference in `jade-devel` that would help solve this problem. The only difference in the serial driver is https://github.com/ros-drivers/nmea_navsat_driver/commit/9fb652d04ba2b71533724b265b360a11ad52d679#diff-0823214bc294dc12622094332dcce72f 
@kmikaz51 Are you sure the node is still running when looking at the topic? (Use rosnode list). An exception might have occured.

evenator, 2022-02-06:
Closing this due to lack of activity.

closed 2022-02-06 by evenator