ros-drivers/nmea_navsat_driver #48: nmea_serial_driver ,no data when I run rostopic echo /fix

Phadadev, 2018-06-14:
I received these message after I run `rosrun nmea_navsat_driver nmea_serial_driver`

> [WARN] [1528961811.402985]: Received a sentence with an invalid checksum. Sentence was: "\x00\x00\x00\xbey\xf2\x00\x0f'\x00\x00\x12\x19&;\x00\x00\x00\x00\x00\x00\x00\x00\xf4\xa1\xb5b\x01\x04\x12\x00P\xf9;\x16\x1a\x00\x0f'\x1a\x00\x0f'\x0f'\x0f'\x0f'\xf3b\xb5b"

> 

And then I don't find anything after I run `rostopic echo -c /fix`

evenator, 2018-08-11:
It looks like you're running your GPS device in some kind of binary mode. This package only supports [NMEA string](http://www.gpsinformation.org/dale/nmea.htm) output.

closed 2018-08-11 by evenator

TSRyan, 2018-09-14:
I was/am dealing with a similar issue.  Originally, I was getting output like that above, so I tried looking at the output directly from the port (`cat /dev/ttyUSB0`), but the process returned immediately with no output.  Next, I tried `screen /dev/ttyUSB 57600` and received readable output:
```
$GPGSV,3,3,09,36,,,*75
$GPRMC,155741.400,A,4129.2735,N,07538.4695,W,0.60,170.45,140918,,,A*75
$GPVTG,170.45,T,,M,0.60,N,1.11,K,A*3D
$GPGGA,155741.600,4129.2735,N,07538.4694,W,1,5,2.10,486.1,M,-33.8,M,,*68
$GPGLL,4129.2735,N,07538.4694,W,155741.600,A,A*45
$GPGSA,A,3,06,12,02,17,09,,,,,,,,2.31,2.10,0.97*07
```
After using the `screen` command, I am able to see the same output from `cat /dev/ttyUSB0`.  Clearly, the baud rate was the issue there.  I confirmed this by reattaching the USB (reinitializing the port), then issuing:
```
stty -F /dev/ttyUSB0 57600
cat /dev/ttyUSB0
```
The output was correct (as above).

Having confirmed proper setup of the serial port, I once again tried `nmea_serial_driver`:
```
rosrun nmea_navsat_driver nmea_serial_driver _baud:=57600
```
 The serial output is still not parsed correctly, though:
```
[WARN] [1536941034.749468]: Received a sentence with an invalid checksum. Sentence was: 'T,,M,0.35,N,0.65,K,A*3F'
```
After grabbing a random end portion of a NMEA sentence, the process hangs.

I suspect there are other aspects of the serial port that aren't properly configured or are reconfigured by `nmea_serial_driver` (possibly inadvertently).  I will investigate further and open a separate issue if warranted.  Otherwise, it suffices to say that serial port settings are the issue with output like the OP.