ros-drivers/nmea_navsat_driver #50: IOError: [Errno 110] Connection timed out

andrestoga, 2018-07-16:
Hi,

I'm getting the following issue while running the nmea_serial_driver on Kinetic:

```
process[nmea_serial_driver-2]: started with pid [16393]
Traceback (most recent call last):
  File "/home/ucmrobotics/ros_rapid_usda/src/nmea_navsat_driver/scripts/nmea_serial_driver", line 49, in <module>
    GPS = serial.Serial(port=serial_port, baudrate=serial_baud, timeout=20)
  File "/home/ucmrobotics/.local/lib/python2.7/site-packages/serial/serialutil.py", line 240, in __init__
    self.open()
  File "/home/ucmrobotics/.local/lib/python2.7/site-packages/serial/serialposix.py", line 286, in open
    self._update_dtr_state()
  File "/home/ucmrobotics/.local/lib/python2.7/site-packages/serial/serialposix.py", line 634, in _update_dtr_state
    fcntl.ioctl(self.fd, TIOCMBIS, TIOCM_DTR_str)
IOError: [Errno 110] Connection timed out
[nmea_serial_driver-2] process has died [pid 16393, exit code 1, cmd /home/ucmrobotics/ros_rapid_usda/src/nmea_navsat_driver/scripts/nmea_serial_driver fix:=fix __name:=nmea_serial_driver __log:=/home/ucmrobotics/.ros/log/501f3746-889b-11e8-b1b5-035c19e54ecc/nmea_serial_driver-2.log].
log file: /home/ucmrobotics/.ros/log/501f3746-889b-11e8-b1b5-035c19e54ecc/nmea_serial_driver-2*.log
```
but if I run on it on Indigo, it works. Does anybody know what is happening?

Best,

evenator, 2018-07-30:
Are you running it with the same hardware and operating system in both cases? Can you provide some more details, such as the output of `uname -a` on your machine?

ajay1606, 2020-01-09:
@evenator 
AM also getting a similar error when I run the package. Could you please help me to fix this issue ?
[nmea_serial_driver_node-2] process has died [pid 16731, exit code 1, cmd /home/intern/nmea_ws/src/nmea_navsat_driver-master/scripts/nmea_serial_driver __name:=nmea_serial_driver_node __log:=/home/intern/.ros/log/10f3185e-329e-11ea-9ca4-e0d55eb869bd/nmea_serial_driver_node-2.log].
log file: /home/intern/.ros/log/10f3185e-329e-11ea-9ca4-e0d55eb869bd/nmea_serial_driver_node-2*.log
Regards,
Ajay

evenator, 2020-01-11:
Hi @ajay1606. Can you please post the log from when you encountered the error? If you haven't cleaned your ROS logs, it should still be at `/home/intern/.ros/log/10f3185e-329e-11ea-9ca4-e0d55eb869bd/nmea_serial_driver_node-2.log`. That should help diagnose the issue.

ajay1606, 2020-01-13:
Hello @evenator, Thank you so much for your quick response.

- The launch file used to run the current nmea_serial_driver node
roslaunch nmea_navsat_driver nmea_serial_driver.launch

- The log from when you encountered the error. 
[rospy.client][INFO] 2020-01-09 14:08:26,790: init_node, name[/nmea_serial_driver_node], pid[16731]
[xmlrpc][INFO] 2020-01-09 14:08:26,790: XML-RPC server binding to 127.0.0.1:0
[xmlrpc][INFO] 2020-01-09 14:08:26,791: Started XML-RPC server [http://localhost:46641/]
[rospy.init][INFO] 2020-01-09 14:08:26,791: ROS Slave URI: [http://localhost:46641/]
[rospy.impl.masterslave][INFO] 2020-01-09 14:08:26,791: _ready: http://localhost:46641/
[rospy.registration][INFO] 2020-01-09 14:08:26,791: Registering with master node http://localhost:11311
[xmlrpc][INFO] 2020-01-09 14:08:26,791: xml rpc node: starting XML-RPC server
[rospy.init][INFO] 2020-01-09 14:08:26,891: registered with master
[rospy.rosout][INFO] 2020-01-09 14:08:26,892: initializing /rosout core topic
[rospy.rosout][INFO] 2020-01-09 14:08:26,897: connected to core topic /rosout
[rospy.simtime][INFO] 2020-01-09 14:08:26,900: /use_sim_time is not set, will not subscribe to simulated time [/clock] topic
[rospy.core][INFO] 2020-01-09 14:08:27,008: signal_shutdown [atexit]
[rospy.impl.masterslave][INFO] 2020-01-09 14:08:27,009: atexit

- GNSS device model: MRP2000 (https://www.synerex.kr/)

- uname -a
Linux Dgist 4.15.0-74-generic #83~16.04.1-Ubuntu SMP Wed Dec 18 04:56:23 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

- echo $ROS_DISTRO
Kinetic

- rosversion nmea_navsat_driver: 0.5.0

Thank you so much in advance

evenator, 2020-01-14:
@ajay1606 Before we start digging into the driver, can you verify that you can connect to the device using a terminal program such as `miniterm.py` and verify that the device port and baud rate in the launch file are correct?

evenator, 2020-06-13:
I'm closing this out, as the original reporter has not verified that their device and serial connection are working properly.

closed 2020-06-13 by evenator