ros-drivers/nmea_navsat_driver #51: Suggest handling exceptions when killing node

shahvi, 2018-08-02:
Hi,
Currently when we kill the node, it throws the following exceptions / errors.  Suggest handling these so that the node closes gracefully.
```
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/nmea_navsat_driver/nmea_serial_driver", line 52, in <module>
    data = GPS.readline().strip()
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 511, in read
    raise SerialException('read failed: %s' % (e,))
serial.serialutil.SerialException: read failed: (4, 'Interrupted system call')
```

evenator, 2018-08-11:
Hi @shahvi. Thanks for pointing this out. I don't have a lot of time to maintain this package, but I'd welcome a PR from anyone who has time to fix it.

evenator, 2018-08-11:
I just noticed that this was already reported in #47, so please continue discussion on that ticket.

closed 2018-08-11 by evenator