ros-drivers/nmea_navsat_driver #53: Issue echoing or subscribing to /fix topic with Emlid Reach GPS

popenc, 2018-09-13:
I'm encountering an issue that I'm wondering if anyone else has noticed or could sanity check for me. It seems that I can't echo or subscribe to the /fix topic only when I upgrade my GPS's software. I've posted this to the GPS's forums but haven't heard anything.

**Setup:**

 - OS - Ubuntu 14.04
 - More OS info - Linux CPR-J100-0057 3.16.0-77-generic #99~14.04.1-Ubuntu SMP Tue Jun 28 19:17:10 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
 - ROS - Indigo 1.11.20
 - GPS Package - nmea_navsat_driver 0.4.2
 - GPS unit - Emlid Reach RS, ReachView app 2.11 or greater (I have 2.14 at the moment) plugged into computer with USB.

**Issue:**

When I run the nmea_navsat_driver's serial node, I can see the /fix topic with "rostopic list," and I can see it's being published by the serial node with "rostopic info /fix," but when I try to echo the messages or subscribe to them in another node, it hangs up indefinitely.

**Workaround:**

Oddly, if I downgrade Emlid's ReachView application to version 2.10 or lower, this problem resolves itself and I'm able to echo and subscribe to the /fix topic. If I upgrade to 2.11+, the issue comes right back.

**Conclusion:**

Uhh, I'm not really sure where this problem lives, and I've posted over at the Emlid forums (https://community.emlid.com/t/reachview-2-11-and-ros-topics/11152) and was in the middle of posting to the ROS forums when I thought it might be more efficient to post it here as an issue. I mostly wanted to see if anyone has encountered this, or if someone would be bored enough to try and replicate the issue. Also, I want to note that I'm a bit tied to the old Ubuntu and ROS versions because the robot we're using came with them and their firmware is a little too tied to ROS upgrade without breaking stuff.

gavanderhoorn, 2018-09-13:
Cross-post of [nmea_navsat_driver ROS package and Emlid Reach GPS units](https://answers.ros.org/question/303295) on ROS Answers.

popenc, 2018-09-13:
Sorry about that, I'll close it here.

closed 2018-09-13 by popenc