ros-drivers/nmea_navsat_driver #54: Release with new version

roehling, 2018-11-14:
The current version, 0.5.0, is more than three years old. There has been a number of pull requests and bug fixes since then, so it would be nice to see those in the ROS distributions...

evenator, 2018-12-18:
Sorry for the delayed response. I'll work on getting a release out soon (probably over the holidays).

evenator, 2018-12-29:
I've back-ported the following PRs to `kinetic-devel` and `melodic-devel`:

- #21 
- #22 
- #38 
- #28 
- #52 
- #32 
- #30/#58

I'll bloom these shortly as version 0.5.1 for kinetic and melodic. I do not plan to update indigo, which is still on 0.4.x.

I omitted the following PRs because of API/behavior changes that might break existing assumptions:

- #33/#57 (Includes a TF API change)
- #36 (Changes how altitude is calculated)
- #46 (Changes how covariance is calculated)
- #25 (Adds a new topic for heading) and #59 (bugfix for #25)

These PRs will be included in version 0.6.0 for ROS Noetic this spring.

evenator, 2018-12-30:
Version 0.5.1 bloomed for kinetic and melodic.

closed 2018-12-30 by evenator