ros-drivers/nmea_navsat_driver #61: Ros2 support

JaviBecerra, 2019-02-21:
Hi,
We have been using the driver with ros (1) and are planning to migrate to ros2 (crystal), but it seems that nmea_navsat_driver is not supported yet (or any other gps/nmea driver by the way...).
Are there any plans to migrate it to ros2?

Not sure of the effort required to make the migration; we might give it a try if you can provide some guidance as to where we should be starting/what needs to be done.
Regards,
Javier

evenator, 2019-02-24:
This driver is pretty much in maintenance mode, and I don't have any plans to port it to ROS 2.

I don't personally use ROS 2, so I don't know how difficult it would be to support both ROS 1 and 2 in the same package. If you or other users have a good solution to that, I'd be happy to merge it into the package.

klintan, 2019-03-05:
Hey, 
I started the effort, in a branch on my fork here https://github.com/klintan/nmea_navsat_driver/tree/ros2 (some work remaining, but not too much). Perhaps @evenator if you could just create a ros2 branch in your repo I can do a PR, I think keeping it in a separate branch for the foreseeable future is a good idea. It seems that is also how other ros1/ros2 projects are doing it (image_pipeline, image_common etc).

Let me know if there is anything I can do to ease the process.

JaviBecerra, 2019-03-05:
@klintan, we can check it out when you have a working version. We can also give you a hand if you hand us some info on the tasks that have to be done yet.
Thank you both @klintan and @evenator for your effort.

klintan, 2019-03-06:
@JaviBecerra ok so now at least the `nmea_serial_driver` is spitting out some nmea strings, and there are some topics if you run `ros2 topic list`. 

However haven't tested the other nodes yet and still needs some more work (haven't gotten a GPS fix yet so the haven't been able to confirm the messages coming out. Also there are still some other bugs, so if you are in a hurry (or just curious) you could try it out, if not I'll just ping/edit this comment as I resolve some of it.

evenator, 2019-03-06:
It looks like you've made some exciting progress! I'll add a `ros2` branch to hold this work soon. (I'm in the middle of moving, and my ROS PC is in a box right now.)

JaviBecerra, 2019-03-07:
Hi @klintan,
We have tried to get the ros2 node working but are facing a couple problems, probably due to our lack of experience installing ros2 packages...
First we get an error on launch/nmea_serial_driver_launch.py, we had to replace on line 36 
```params="config.yaml")```
with
```params=["config.yaml"])```
or we get a TypeError.
Once that is solved, we get a message stating that the libexec directory does not exist.
We installed the package with command ```colcon build --symlink-install``` without error, but it seems that the nodes are not being installed properly (```ros2 pkg executables | grep nmea``` comes out empty.)
Any idea of what we may be doing wrong?
We are running Ros2 crystal on Ubuntu Bionic.

klintan, 2019-03-07:
Good catch! thanks for testing it out, I'll take a look at all these issues. I hit some issues when creating the custom message nmea_msg/Sentence which delayed stuff. 

In the meantime it should be possible to start the serial node using `python3 nmea_navsat_driver/nodes/nmea_serial_driver.py` or similar.

@evenator no rush, focus on the moving :)

JaviBecerra, 2019-03-08:
Hi @klintan, we tried your latest changes using a serial GPS and everything is working fine.
Thanks!

klintan, 2019-03-16:
@evenator hope your move went well, just a gentle nudge to see if you have time to create a ros2 branch so that I can create a PR for the ROS2 compatibility? :)

evenator, 2019-03-19:
I've created the `ros2` branch. Go ahead and open a PR for your changes.

klintan, 2019-03-20:
Thanks :) PR now here https://github.com/ros-drivers/nmea_navsat_driver/pull/64

evenator, 2019-03-23:
@klintan Thanks for helping out with this! I looked over your PR. I think if we can resolve some of the big questions about moving files around and minimizing divergence, we'll be able to merge it in.

closed 2019-05-08 by evenator