ros-drivers/nmea_navsat_driver #62: Fix missing runtime dependency

evenator, 2019-02-28:
When building the package and running the driver with minimal dependencies installed, the following Python error appears because the dependency on `tf` is not declared in `package.xml`:
```
Traceback (most recent call last):
  File "/home/ed/code/nmea_navsat_driver_ws/install/lib/nmea_navsat_driver/nmea_serial_driver", line 39, in <module>
    import libnmea_navsat_driver.driver
  File "/home/ed/code/nmea_navsat_driver_ws/install/lib/python2.7/dist-packages/libnmea_navsat_driver/driver.py", line 39, in <module>
    from tf.transformations import quaternion_from_euler
ImportError: No module named tf.transformations
```

anubhav1772, 2019-08-14:
Try installing the module using` sudo apt-get install ros-<ros-distro>-tf-transformations`.

evenator, 2019-08-14:
That's a good workaround until we can fix the package dependencies so it's installed automatically.

PR #102 (2020-06-13): Add missing dependency on tf
When building the package and running the driver with minimal dependencies installed, a Python error appears because the dependency on tf is not declared in package.xml. Declare that dependency.

Fixes #62.
commit 32fbd621 (2020-06-13): Add missing dependency on tf (#102)

When building the package and running the driver with minimal dependencies installed, a Python error appears because the dependency on tf is not declared in package.xml. Declare that dependency.

Fixes #62.
closed 2020-06-13 by evenator