ros-drivers/nmea_navsat_driver #86: Can't parse alt_std_dev from GST correctly

NearLinHere, 2019-12-12:
Hi all,
Thank you for making this great parse driver. The code is clean and works well.

Recently, I have been trying to retrieve the information in GST. I found that the value of alt_std_dev is always nan, and I added some log to figure out what happened. 

Here is where I put the log in parse_nmea_sentence() in parser.py:

    if sentence_type == "RMC":
        parsed_sentence["utc_time"] = convert_time_rmc(fields[9], fields[1])

    # log for check GST
    if sentence_type == "GST":
        rospy.logwarn(
                "nmea_sentence: %s" % nmea_sentence)
        rospy.logwarn(
                "parsed_sentence: %s" % parsed_sentence)
        rospy.logwarn(
                "fields: %s" % fields)

    return {sentence_type: parsed_sentence}

And following is the output message:

```
[WARN] [1576155838.636761]: nmea_sentence: $GNGST,130358.60,36,,,,0.046,0.041,0.057*43
[WARN] [1576155838.638862]: parsed_sentence: {'semi_major_ellipse_std_dev': nan, 'alt_std_dev': nan, 'lon_std_dev': 0.041, 'semi_major_orientation': nan, 'ranges_std_dev': 36.0, 'semi_minor_ellipse_std_dev': nan, 'lat_std_dev': 0.046, 'utc_time': (1576155838, 600000000)}
[WARN] [1576155838.641648]: fields: ['$GNGST', '130358.60', '36', '', '', '', '0.046', '0.041', '0.057*43']
```

It seems like the [*CheckSum] value should be removed before the parse to get correct alt_std_dev. Or did I do something wrong? Any advice would be appreciated.

evenator, 2019-12-16:
Thanks for pointing out this bug in the parser. I haven't tried to recreate the error yet, but I took a stab at fixing it. Can you try out the change in pull request #87?

NearLinHere, 2019-12-18:
Hi @evenator,
Thank you for fix this so quickly. However, I don't have time to test it recently.  :(

PR #87 (2020-06-13): Use a regex to split tokens in parser
NMEA uses a * character to delimit the beginning of the regex, so use a regex that matches * and , as the field delimiter.

Fixes #86
closed 2020-06-13 by evenator