ros-drivers/nmea_navsat_driver #89: Nothing published to /fix

rgov, 2020-01-16:
I am ROS newbie trying to use the `nmea_socket_driver` to receive UDP packets containing NMEA sentences and re-publish parsed NavSatFix messages. I'm using Melodic.

I have the node running and I even augmented the script with print statements to show that it is receiving NMEA sentences.

```
ros-nodes_1   | [WARN] [1579139206.660089]: Receiving!
ros-nodes_1   | [WARN] [1579139207.765352]: Received '$GPGGA,014647.000,4131.46214,N,07040.11177,W,0,03,99.0,014.32,M,-34.4,M,,*66\r\n$GPVTG,0.0,T,,M,0.0,N,0.0,K,N*02\r\n$GPRMC,014647.000,V,4131.46214,N,07040.11177,W,0.0,0.0,160120,,,N*60\r\n'
ros-nodes_1   | [WARN] [1579139207.774498]: Receiving!
ros-nodes_1   | [WARN] [1579139214.165529]: Received '$GPGGA,014654.000,4131.46214,N,07040.11177,W,0,03,99.0,014.32,M,-34.4,M,,*64\r\n$GPVTG,0.0,T,,M,0.0,N,0.0,K,N*02\r\n$GPRMC,014654.000,V,4131.46214,N,07040.11177,W,0.0,0.0,160120,,,N*62\r\n'
ros-nodes_1   | [WARN] [1579139214.173671]: Receiving!
ros-nodes_1   | [WARN] [1579139220.163735]: Received '$GPGGA,014700.000,4131.46214,N,07040.11177,W,0,03,99.0,014.32,M,-34.4,M,,*64\r\n$GPVTG,0.0,T,,M,0.0,N,0.0,K,N*02\r\n$GPRMC,014700.000,V,4131.46214,N,07040.11177,W,0.0,0.0,160120,,,N*62\r\n'
ros-nodes_1   | [WARN] [1579139220.171450]: Receiving!
```

However, if I use `rostopic echo /fix` I never see any messages dumped, which suggests to me that something is breaking down. What can I do to diagnose this?

evenator, 2020-01-19:
Thanks for reporting this and making a pull request with the fix!

rgov, 2020-01-19:
I would recommend additional logging for this. I'm not sure why I wasn't seeing any warning. For instance, if the line is blank, then you will get an error about the checksum being invalid. Why wasn't it triggering the same way on a line ending with `\r`?

evenator, 2020-01-19:
Here's why the checksum passes:
```Python
def check_nmea_checksum(nmea_sentence):
    split_sentence = nmea_sentence.split('*')
    if len(split_sentence) != 2:
        # No checksum bytes were found... improperly formatted/incomplete NMEA data?
        return False
    transmitted_checksum = split_sentence[1].strip() # This is where the `\r` is removed.

    # Remove the $ at the front
    data_to_checksum = split_sentence[0][1:]
    checksum = 0
    for c in data_to_checksum:
        checksum ^= ord(c)

    return ("%02X" % checksum) == transmitted_checksum.upper()
```
For the first line in your example above:
```
nmea_sentence =
    '$GPGGA,014647.000,4131.46214,N,07040.11177,W,0,03,99.0,014.32,M,-34.4,M,,*66\r'
split_sentence = [
    '$GPGGA,014647.000,4131.46214,N,07040.11177,W,0,03,99.0,014.32,M,-34.4,M,,',
    '66\r']
len(split_sentence) = 2
transmitted_checksum = '66' # This is where the `\r` is removed.
checksum = 66
("%02X" % checksum) = '66'
# checksum passes
```
On the other hand, the validation regex in the parser rejects a string that doesn't end with a hex number:
```
r'(^\$GP|^\$GN|^\$GL|^\$IN).*\*[0-9A-Fa-f]{2}$'
```
I think the parser should strip whitespace like `check_nmea_checksum`  does.

evenator, 2020-06-13:
Fixed by #98

closed 2020-06-13 by evenator