ros-drivers/nmea_navsat_driver #94: "REceived a Sentence with Invalid Checksum sentence was : " "

sdey76, 2020-02-02:
Hii
Am getting below error with NEO 6M GPS MODULE, using ROS kinetic 
rosrun nmea_navsat_driver nmea_serial_driver _port:=/dev/ttyUSB0 _baud:=38400 
KIndly help. Thanks a lot
[image]

evenator, 2020-02-02:
Your device appears to be sending sentences that contain no text. I suspect the baud rate of the driver does not match the baud rate of the device.

sdey76, 2020-02-02:
Thanks, I have set different baudrates, 4800 to 115200 also changed the same in the nmea serial driver.py

sdey76, 2020-02-03:
Hii
Can you suggest a gps module ( proven module here) that can interface with my ROS autonomous indoor lidar based robot , robot computer is a Jetson Nano, to do outdoor navigation . Help appreciated please. Thanks  a lot.

lillefyr, 2020-02-03:
I can not find an example in the doc of a correct call ( https://wiki.ros.org/nmea_navsat_driver ), however, are you sure the parameter is passed this way?

_port:=/dev/ttyUSB0     I think it should be   **-port=/dev/ttyUSB0**

and the same for same for baud!

You try to change the port and the baud, BUT, because your parameters are specified wrong, you always use port=/dev/ttyUSB0 and baud=4800 because that is the default.

sdey76, 2020-02-04:
Resolved

Rx Tx - Arduino UNO used and not TTL USB Convertor
, with rosserial arduino

[image]

closed 2020-02-04 by sdey76