ros-drivers/nmea_navsat_driver #96: ublox gps LEA-6T received a sentencce with an invalid checksum. sentence was

abdo-robotic, 2020-03-09:
hello, I am using a Ublox lea 6T GPS for robot navigation. to connect the GPS with raspberry pi I've used TTL USB. after nmea_navsat_driver package installation and running: rosrun nmea_navsat_driver nmea_serial_driver _port:=/dev/ttyUSB0 _baud:=9600 I got a warning msg: Received a sentence with an invalid checksum. The sentence was:'\x\xc\xefkl and when I run: rostopic list /fix /heading /rosout /time_reference /vel when I want to display topics I got no data

evenator, 2020-03-10:
It looks like either your baud rate is wrong or your device is not configured to output NMEA sentences.

closed 2020-03-10 by evenator

abdo-robotic, 2020-03-10:
excuse me I am not familiar with NMEA it's my first time with a similar device. can you please explain to me how I can configure my device to output NMEA sentences??

evenator, 2020-03-10:
Please direct questions like this to https://answers.ros.org/questions/. Thanks!

abdo-robotic, 2020-03-10:
thank you, for replying
I checked this link https://www.handheldgroup.com/knowledge-base/enable-or-disable-nmea-datasets-in-u-blox-gps-module/  to enable NMEA datasets by u-center unfortunately when I retry running  nmea_navsat_driver I got the same output, I think there is something missing I wish if there is someone can help