ros-drivers/nmea_navsat_driver #103: Release for ROS Noetic

evenator, 2020-06-13:
Test this package in ROS Noetic (notably, this requires Python3) and release a new version to ROS Noetic.

donkikos, 2020-09-22:
Hey @evenator any updates on this?

evenator, 2020-09-22:
The package was released in Noetic, but there's a known issue with Python 3 (see #111, and potential fix in #112).

donkikos, 2020-09-22:
Thanks for quick response. I was able to clone, compile and run nmea_navsat_driver with 3 changes in different files. I also got stuck for some time on the issue you mentioned and was not able to figure out what that check meant. Thanks for the links, at least it clears out what the if is meant to check. 

So is the package in the ubuntu repos? I was not able to install it with apt. It does not show here too - https://wiki.ros.org/nmea_navsat_driver

evenator, 2020-09-26:
Looks like you're right. I could have sworn I released it for noetic. I'll test it today and bloom a release.

evenator, 2020-09-26:
Release is pending ros/rosdistro#26721

closed 2020-09-26 by evenator

donkikos, 2020-09-26:
Great, thank you @evenator !