ros-drivers/nmea_navsat_driver #104: Bug in parsing multiple types of messages

ybc82, 2020-06-21:
My GPS was publishing GNGGA and GNHDT through UDP at 20 Hz. I was using nmea_socket_driver.py. All GNGGA messages were not successfully parsed because the original string was "$GNGGA ... \r\n", therefore ending with \r after stripping the '\n'. The GNHDT messages ended with \n only and were successfully parsed.

My quick fix was to further strip '\r' in the GNGGA message. I was not sure this is a good fix for the whole driver, so I'm posting an issue here instead of a pull request.

evenator, 2020-06-21:
Hi, can you answer a few questions about your setup for me?

- What version of ROS are you using?
- Did you install this package with apt or download the source code with git, or install it some other way? 
- What is the output of `rosversion nmea_navsat_driver`?

ybc82, 2020-06-22:
> Hi, can you answer a few questions about your setup for me?
> 
> * What version of ROS are you using?
> * Did you install this package with apt or download the source code with git, or install it some other way?
> * What is the output of `rosversion nmea_navsat_driver`?

I just realized that this bug has been fixed recently... Thank you for your reply though!

1. I'm using ROS Kinetic;
2. I downloaded the source code. But wait! I haven't updated the code your code in the past two weeks because I was using my fork. It was fixed in #98. 
3. 0.5.0

closed 2020-06-28 by evenator