ros-drivers/nmea_navsat_driver #105: Error in parsing 6-digit UTC time

philsuematsu, 2020-06-25:
When the received NMEA sentence (from a Garmin 18x GPS unit) is:

`$GPGGA,004646,3350.7509,N,11818.7148,W,1,06,1.3,52.6,M,-35.2,M,,*49`

the six-digit UTC time field value, 004646, is not parsed correctly by parser.py:

`131: nanosecs  =  int(nmea_utc[7:]) *  pow(10, 9  -  len(nmea_utc[7:]))`

and produces the error message:

`[WARN] [1593042694.453486]: Value error, likely due to missing fields in the NMEA message. Error was: invalid literal for int() with base 10: ''. Please report this issue at github.com/ros-drivers/nmea_navsat_driver, including a bag file with the NMEA sentences that caused it.`

ggaessler, 2020-06-25:
The driver assumes that the GGA-format is `HHMMSS.SS`, which holds for nearly all receivers. However it seems like this is not the case for all Garmin receivers. 

A fix would be changing

`nanosecs = int(nmea_utc[7:]) * pow(10, 9 - len(nmea_utc[7:]))`

into

```
nanosecs = 0
if len(nmea_utc) > 7:
    nanosecs = int(nmea_utc[7:]) * pow(10, 9 - len(nmea_utc[7:]))```

evenator, 2020-06-28:
Hi @philsuematsu, can you please test the branch I just added in PR #106 and confirm that it fixes your problem?

evenator, 2020-10-31:
I'm going to close this as fixed.

closed 2020-10-31 by evenator