ros-drivers/nmea_navsat_driver #108: Incorrect Longitude output

osholsonel20, 2020-07-14:
When trying to run the nmea_navsat_driver the output of longitude is not correct.

[image]
[image]

Adding an example sentence around the area:
$GPGGA,202656.032,4400.625198,N,8832.555418,W,1,10,0.01,0.021805,M,0.000000,M,,*7C

Using LGSVL simulator, and when running the output message string into a GPS decoder website, it does seem to have the correct location. However, when running the nmea_navsat_driver the longitude is way off

osholsonel20, 2020-07-14:
Fixed by setting line 60: in parser.py.

return safe_float(field[0:2]) + safe_float(field[2:]) / 60.0

instead of 

return safe_float(field[0:3]) + safe_float(field[3:]) / 60.0

closed 2020-07-14 by osholsonel20

evenator, 2020-07-18:
I've checked several online references for the longitude field of the GGA sentence, and it appears that the NMEA standard expects that the format for longitude be _3 digits_ for degrees longitude followed by the digits for decimal minutes. It looks like your receiver is outputting an incorrect sentence, because it should have a leading zero. `8832.555418` should be `08832.555418`.

I'm glad you found a workaround for your receiver, but I don't think there's an error in this package's GGA parser.

osholsonel20, 2020-07-20:
It was in LGSVL's message output of gps. It was missing a leading zero, and honestly I'd rather put a leading zero in the output vs. messing with the input! Was a good temporary fix, then I realized the message being received was formatted wrong