ros-drivers/nmea_navsat_driver #111: Python error: TypeError: '>' not supported between instances of 'list' and 'int'

Tobias-Fischer, 2020-09-08:
Hiya,

I am getting this error since moving to Noetic:

```
Sep 09 07:26:27 cpr-qut10 ros-start[1494]: TypeError: '>' not supported between instances of 'list' and 'int'
Sep 09 07:26:27 cpr-qut10 ros-start[1494]: [ERROR] [1599600387.886435]: bad callback: <function nmea_sentence_callback at 0x7fa1752e3d30>
Sep 09 07:26:27 cpr-qut10 ros-start[1494]: Traceback (most recent call last):
Sep 09 07:26:27 cpr-qut10 ros-start[1494]:   File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 748, in _invoke_callback
Sep 09 07:26:27 cpr-qut10 ros-start[1494]:     cb(msg, cb_args)
Sep 09 07:26:27 cpr-qut10 ros-start[1494]:   File "/home/administrator/catkin_ws/src/nmea_navsat_driver/src/libnmea_navsat_driver/nodes/nmea_topic_driver.py", line 50, in nmea_sentence_callback
Sep 09 07:26:27 cpr-qut10 ros-start[1494]:     driver.add_sentence(
Sep 09 07:26:27 cpr-qut10 ros-start[1494]:   File "/home/administrator/catkin_ws/src/nmea_navsat_driver/src/libnmea_navsat_driver/driver.py", line 213, in add_sentence
Sep 09 07:26:27 cpr-qut10 ros-start[1494]:     if gps_qual > 0:
Sep 09 07:26:27 cpr-qut10 ros-start[1494]: TypeError: '>' not supported between instances of 'list' and 'int'
```

Any idea why this would be?

Tobias-Fischer, 2020-09-08:
Actually, I think it's a Python2 vs Python3 thing:
On Python2, one can compare a list to an int (whatever that means), whereas on Python3 it fails with above TypeError. I am not sure what this comparison should check though, happy to provide a PR with a fix if someone tells me.

evenator, 2020-09-13:
Can you test out PR #112?

Tobias-Fischer, 2020-09-14:
Looks good. Haven't tried with to get a fix outdoors yet, but at least it doesn't throw an error anymore.

evenator, 2020-09-26:
Fixed by #112

closed 2020-09-26 by evenator