ros-drivers/nmea_navsat_driver #115: UnicodeDecodeError: 'utf-8' codec can't decode byte 0xc6 in position 6: invalid continuation byte

donkikos, 2020-10-09:
The new release for noetic gives error. Trying to use with [GlobalSat BU-353-S4](https://smile.amazon.com/gp/product/B008200LHW/). 

```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.1 LTS
Release:	20.04
Codename:	focal
```

```
$ rosrun nmea_navsat_driver nmea_serial_driver _port:=/dev/gps _baud:=4800
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/nmea_navsat_driver/nmea_serial_driver", line 38, in <module>
    libnmea_navsat_driver.nodes.nmea_serial_driver.main()
  File "/opt/ros/noetic/lib/python3/dist-packages/libnmea_navsat_driver/nodes/nmea_serial_driver.py", line 67, in main
    nmea_str = data.decode('utf-8')
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xc6 in position 6: invalid continuation byte
```

evenator, 2020-10-10:
Thanks for reporting this. I think I know what's happening here. NMEA strings should be ASCII-encoded, but the first sentence from the serial device often contains some garbage data, which is very likely to include bytes outside of the ASCII table (and in this case outside of the utf-8 table even). I'll put up a fix shortly that discards such data so that the node doesn't crash in this case.

evenator, 2020-10-10:
@donkikos Would you mind reviewing PR #116?

donkikos, 2020-10-10:
@evenator will test with the receiver I got. 

I was patching by changing the `decode('utf-8')` with `decode(encoding='utf-8', errors='ignore')` and seems it was working.

evenator, 2020-10-11:
I considered `errors='ignore'` as a solution as well. I decided that it would be better to catch the error and print a warning, because otherwise the driver might silently discard all data (like if the baud rate was set incorrectly).

donkikos, 2020-10-12:
> I considered `errors='ignore'` as a solution as well. I decided that it would be better to catch the error and print a warning, because otherwise the driver might silently discard all data (like if the baud rate was set incorrectly).

I agree. 

I tested the `nmea_serial_driver node`. No crashes as of now.

commit 92a799c6 (2020-10-12): Fix #115 Decoding errors. (#116)

Handle decoding errors when reading from a
serial port. NMEA sentences should be ASCII-
encoded, but if malformed data comes in
from the serial port, it won't decode properly.
If the exception isn't handled, the node will
crash. This commonly happens on startup.

Also a couple of codestyle fixes.
closed 2020-10-12 by evenator