ros-drivers/nmea_navsat_driver #120: Error msg parse and checksum

adamkrawczyk, 2021-01-07:
Hi, I run into a problem with nmea_socket_driver  with the remote device sending data over UDP.

ROS Noetic
Ubuntu 20.04

First:

```
Exception happened during processing of request from ('10.15.20.1', 44265)
Traceback (most recent call last):
  File "/usr/lib/python3.8/socketserver.py", line 316, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/usr/lib/python3.8/socketserver.py", line 347, in process_request
    self.finish_request(request, client_address)
  File "/usr/lib/python3.8/socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/lib/python3.8/socketserver.py", line 720, in __init__
    self.handle()
  File "/opt/ros/noetic/lib/python3/dist-packages/libnmea_navsat_driver/nodes/nmea_socket_driver.py", line 58, in handle
    self.server.driver.add_sentence(line, self.server.frame_id)
  File "/opt/ros/noetic/lib/python3/dist-packages/libnmea_navsat_driver/driver.py", line 163, in add_sentence
    if not check_nmea_checksum(nmea_string):
  File "/opt/ros/noetic/lib/python3/dist-packages/libnmea_navsat_driver/checksum_utils.py", line 44, in check_nmea_checksum
    split_sentence = nmea_sentence.split('*')
TypeError: a bytes-like object is required, not 'str'
```

Second:

```
Exception happened during processing of request from ('10.15.20.1', 50240)
Traceback (most recent call last):
  File "/usr/lib/python3.8/socketserver.py", line 316, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/usr/lib/python3.8/socketserver.py", line 347, in process_request
    self.finish_request(request, client_address)
  File "/usr/lib/python3.8/socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/lib/python3.8/socketserver.py", line 720, in __init__
    self.handle()
  File "/opt/ros/noetic/lib/python3/dist-packages/libnmea_navsat_driver/nodes/nmea_socket_driver.py", line 58, in handle
    self.server.driver.add_sentence(line, self.server.frame_id)
  File "/opt/ros/noetic/lib/python3/dist-packages/libnmea_navsat_driver/driver.py", line 168, in add_sentence
    parsed_sentence = libnmea_navsat_driver.parser.parse_nmea_sentence(
  File "/opt/ros/noetic/lib/python3/dist-packages/libnmea_navsat_driver/parser.py", line 284, in parse_nmea_sentence
    if not re.match(
  File "/usr/lib/python3.8/re.py", line 191, in match
    return _compile(pattern, flags).match(string)
TypeError: cannot use a string pattern on a bytes-like object
```

I made a solution but not sure whether it's appropriate and work with any case could you check this? 

Best, 
Adam

adamkrawczyk, 2021-01-07:
PR made at https://github.com/ros-drivers/nmea_navsat_driver/issues/120#issue-781376279

evenator, 2021-01-09:
Thanks for pointing this out. When I migrated to Python 3 for ROS noetic, I added bytes->str decoding in all the node types but this one. The driver still expects its input to be a `str`, not `bytes`, so your change in #121 fixes the UDP node, but would break the other nodes. Can you try out #122?

adamkrawczyk, 2021-01-12:
I have tried but I got strange error named : 

```
Traceback (most recent call last):
  File "/path/catkin_ws/src/nmea_navsat_driver/scripts/nmea_socket_driver", line 37, in <module>
    libnmea_navsat_driver.nodes.nmea_socket_driver.main()
  File "/path/catkin_ws/src/nmea_navsat_driver/src/libnmea_navsat_driver/nodes/nmea_socket_driver.py", line 105, in main
    server.server_bind()
  File "/usr/lib/python3.8/socketserver.py", line 466, in server_bind
    self.socket.bind(self.server_address)
TypeError: an integer is required (got type str)
```

Not sure why this error occurs.

adamkrawczyk, 2021-01-12:
This solves the issues. Thank you :1st_place_medal:

PR #122 (2021-01-17): Decode lines from the UDP server before passing them to the driver
RosNMEADriver.add_sentence() expects the first argument to be a str, not a bytes object.
This was properly decoded in the other node types, but not in the UDP socket server.

Should fix #120
closed 2021-01-12 by adamkrawczyk