ros-drivers/nmea_navsat_driver #131: ROS2 Foxy Status?

mr337, 2021-07-30:
We are making the transition from ROS1 to ROS2. It looks like there was some work getting this package compatible with ROS2. https://github.com/ros-drivers/nmea_navsat_driver/issues/61

Checking [ROS packages](https://index.ros.org/p/nmea_navsat_driver/github-ros-drivers-nmea_navsat_driver/#noetic) and apt not seeing a Foxy version. :(

Does anyone know the blocker to updating this to support Foxy? Also happy to lend a hand as part of the upgrade process.

maxpolzin, 2021-11-03:
What needs to be done to officially release the driver fro ROS2? I am happy to put some effort into this.

evenator, 2022-02-06:
I don't know how well the ROS 2 branch works (or doesn't work) in Foxy. The first step would be for someone to install the package from source in foxy and see what needs to be updated or fixed.

mr-church, 2023-04-18:
I am trying to get this driver working in Foxy with a Garmin 18x sensor. Any updates on this issue that I should take note of?

evenator, 2023-09-01:
Closing this as Foxy is EOL as of June 20, 2023

closed 2023-09-01 by evenator