ros-drivers/nmea_navsat_driver #140: I want to install from source,when I run catkin_make,this occured "Could not find the required component 'catkin_virtualenv'.", how can I solve this problem ?

fengyikawhi, 2022-01-21:


evenator, 2022-01-21:
Can you provide some more info, such as your ROS version, OS version, and the full process you've attempted to follow to install from source?

fengyikawhi, 2022-01-21:
Thanks for your apply, Ubuntu20.04, ROS noetic, 
process:
 mkdir -p ~/catkin_ws/src
cd ~/catkin_ws/src
git clone https://github.com/ros-drivers/nmea_navsat_driver.git
cd ~/catkin_ws
catkin_make

evenator, 2022-01-22:
Ensure you have installed and initialized [rosdep](http://wiki.ros.org/rosdep), then use it to install the dependencies:
```
cd ~/catkin_ws/src
rosdep install --from-paths nmea_navsat_driver --ignore-src
```

fengyikawhi, 2022-01-24:
> Ensure you have installed and initialized [rosdep](http://wiki.ros.org/rosdep), then use it to install the dependencies:
> 
> ```
> cd ~/catkin_ws/src
> rosdep install --from-paths nmea_navsat_driver --ignore-src
> ```

Excellent! I solved the problem, thanks a lot!

closed 2022-02-06 by evenator