ros-drivers/nmea_navsat_driver #141: ROS2 FOXY nmea_socket_driver timed out.

Sylvania2, 2022-01-21:
I am unable to connect to rtkrcv output with nmea_socket_driver.. what am i missing?
I edited ip and port in nmea_socket_driver.py!  

when i run: ros2 run nmea_navsat_driver nmea_socket_driver
i get:
[nmea_navsat_driver]:  Using parameters ip 192.168.1.147 port 12346 buffer_size 4096 timeout_sec 2
Caught exception socket.error during recvfrom: timed out
Caught exception socket.error during recvfrom: timed out
...

Putty connected to rtkrcv server 192.168.1.147/ 12346 (raw) is showing nmea data:
$GPRMC,115234.60,A,5728.5732742,N,01012.0597855,E,0.04,75.06,210122,0.0,E,R,V*63
$GPGGA,115234.60,5728.5732742,N,01012.0597855,E,4,17,1.0,30.197,M,38.131,M,0.0,0000*79

(branch is ros2)

Best regards

Sylvania2, 2022-01-24:
Found a solution:

in nmea_socket_driver.py:

            # Create a socket
            socket_ = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)

            # Bind the socket to the port
            socket_.bind((local_ip, local_port))

change to:

            # Create a socket
            socket_ = socket.socket()

            # Bind the socket to the port
            socket_.connect((local_ip, local_port))

Edit Config and Launch file to use ip/port and socket.
Run:
ros2 launch nmea_navsat_driver nmea_socket_driver.launch.py
Now i receive data from RTKLIB's rtkrcv program setup with tcpsvr output, and ros2 topic echo /fix shows data:

header:
  stamp:
    sec: 1643017694
    nanosec: 130581854
  frame_id: gps
status:
  status: 2
  service: 1
latitude: 57.47622123833333
longitude: 10.200996425
altitude: 68.328
position_covariance:
- 0.0004
- 0.0
- 0.0
- 0.0
- 0.0004
- 0.0
- 0.0
- 0.0
- 0.0064
position_covariance_type: 1

Best regards

evenator, 2022-01-24:
Your device is using TCP. This driver currently only supports UDP. I would be happy to consider a pull request to enable support for both protocols.

Sylvania2, 2022-01-25:
@evenator  Correct, RTKLIB's rtkrcv output is a TCP server.. request: protocol setting in config.file ?

closed 2022-01-30 by Sylvania2