ros-drivers/nmea_navsat_driver #153: connect refused

Liu2022, 2022-09-08:
Hello, everyone.

When I try to use nmea_navsat_driver to convert my GPS data into /nmea_msgs/Sentence, it does not work well.

My command is rosrun nmea_navsat_driver nmea_topic_serial_reader _port:=/dev/ttyUSB0 _baud:=115200. And it turns out Connection refused. Details are shown before.

 File "/opt/ros/noetic/lib/nmea_navsat_driver/nmea_topic_serial_reader", line 38, in <module>
    libnmea_navsat_driver.nodes.nmea_topic_serial_reader.main()
  File "/opt/ros/noetic/lib/python3/dist-packages/libnmea_navsat_driver/nodes/nmea_topic_serial_reader.py", line 59, in main
    rospy.init_node('nmea_topic_serial_reader')
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/client.py", line 326, in init_node
    _init_node_params(argv, name)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/client.py", line 186, in _init_node_params
    set_param(rosgraph.names.PRIV_NAME + param_name, param_value)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/client.py", line 539, in set_param
    _param_server[param_name] = param_value #MasterProxy does all the magic for us
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/msproxy.py", line 136, in __setitem__
    self.target.setParam(rospy.names.get_caller_id(), resolved_key, val)
  File "/usr/lib/python3.8/xmlrpc/client.py", line 1109, in __call__
    return self.__send(self.__name, args)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/core.py", line 669, in _ServerProxy__request
    return xmlrpcclient.ServerProxy._ServerProxy__request(
  File "/usr/lib/python3.8/xmlrpc/client.py", line 1450, in __request
    response = self.__transport.request(
  File "/usr/lib/python3.8/xmlrpc/client.py", line 1153, in request
    return self.single_request(host, handler, request_body, verbose)
  File "/usr/lib/python3.8/xmlrpc/client.py", line 1165, in single_request
    http_conn = self.send_request(host, handler, request_body, verbose)
  File "/usr/lib/python3.8/xmlrpc/client.py", line 1278, in send_request
    self.send_content(connection, request_body)
  File "/usr/lib/python3.8/xmlrpc/client.py", line 1308, in send_content
    connection.endheaders(request_body)
  File "/usr/lib/python3.8/http/client.py", line 1251, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.8/http/client.py", line 1011, in _send_output
    self.send(msg)
  File "/usr/lib/python3.8/http/client.py", line 951, in send
    self.connect()
  File "/usr/lib/python3.8/http/client.py", line 922, in connect
    self.sock = self._create_connection(
  File "/usr/lib/python3.8/socket.py", line 808, in create_connection
    raise err
  File "/usr/lib/python3.8/socket.py", line 796, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused

I hope someone can give me a hand and I am very appreciate for it.

Thank you!

zhangjinwei786, 2022-10-07:
Did you run roscore beforehead?

evenator, 2022-10-16:
As @zhangjinwei786 points out, this error message indicates that the node can't connect to a ROS core. You either need to run `roscore` or launch the node with a launch file and `roslaunch`, which will automatically start a ROS core if one isn't running.

closed 2022-10-16 by evenator