ros-drivers/nmea_navsat_driver #155: ModuleNotFoundError: No module named 'serial'

aristow1, 2022-11-03:
Hi im using Ubuntu 18.04 and ros melodic. I tried launching the "roslaunch nmea_navsat_driver nmea_serial_driver.launch" file and i get this error : ModuleNotFoundError: No module named 'serial'.
i tried installing serial with : sudo apt-get install python-serial and with pip install serial and i have everything installed : Reading package lists... Done
Building dependency tree       
Reading state information... Done
python-serial is already the newest version (3.4-2).
The following package was automatically installed and is no longer required:
  linux-hwe-5.4-headers-5.4.0-84
Use 'sudo apt autoremove' to remove it.
 and i still get the same error. i dont know how to solve it. any idea ?

Joy-Zhang-0303, 2022-11-05:
Hello, I'm also using this package. Everything is also. My Python is 2.7 . Could you show the Python installed path ?

aristow1, 2022-11-08:
HI my python installation path is /usr/bin/python. im also using python 2.7. i also use melodic-devel

aristow1, 2022-11-08:
how do i check the name of the serial port the gps is connected to. im using a COM port ?

aristow1, 2022-11-14:
i will try this tomorrow and let you know if it works. thank you :)

aristow1, 2022-11-16:
i was using python 3 to build the package and that was the Problem. after using python2.7 to build + the solution of @evenator the Problem was solved. Thanks !

closed 2022-11-16 by aristow1