ros-drivers/nmea_navsat_driver #163: Wrong heading from VTG message

gislers, 2023-04-01:
In the VTG message, the field `true_course` is parsed as `safe_float`, but it should be `convert_deg_to_rads` as in the RMC message. NMEA reference, where field 1 (Track made good) is in degrees: https://receiverhelp.trimble.com/alloy-gnss/en-us/NMEA-0183messages_VTG.html

I'm extracting robot orientation using the direction of the published velocity vector in the /vel topic, but the orientation is jumping wildly. 

Replacing `safe_float` with `convert_deg_to_rads` fixed the issue for me:
https://github.com/ros-drivers/nmea_navsat_driver/blob/b168f499349e5f26e3aed9982deb44ed2f41f8e2/src/libnmea_navsat_driver/parser.py#L265

evenator, 2023-04-02:
Thanks for finding this. If you have time, I would appreciate a pull request with the fix.

gislers, 2023-04-03:
Sure, sorry for not providing it right away... :-)

closed 2023-04-04 by evenator