ros-drivers/nmea_navsat_driver #168: Value error, likely due to missing fields in the NMEA message. Error was: 'utf-8' codec can't decode

ashBabu, 2023-11-01:
On ROS humble, I get the following when I run `ros2 launch theia_localization nmea_serial_driver.launch.py`

`[nmea_serial_driver-1] [INFO] [1698850685.891022741] [nmea_navsat_driver]: Successfully connected to /dev/ttyACM0 at 4800.
[nmea_serial_driver-1] [WARN] [1698850686.079509490] [nmea_navsat_driver]: Value error, likely due to missing fields in the NMEA message. Error was: 'utf-8' codec can't decode byte 0xb5 in position 0: invalid start byte.`

brjthomas, 2023-11-02:
Did you figure this out yet?

ashBabu, 2023-11-02:
No

evenator, 2023-12-26:
`0xb5` is not a valid character values for ASCII (or even UTF-8). NMEA sentences may only contain ASCII characters. If you're getting this byte from your device, it means one of three things:

1. Your device outputs invalid data containing non-ASCII characters, and therefore is not producing valid NMEA sentences. (Unlikely.)
2. Your serial port's baud rate is not set the same as your device, which is causing the serial port to produce garbage data.
3. This is a transient error that only occurs when the serial port first connects, in which case it can be ignored.

closed 2023-12-26 by evenator