ros-drivers/nmea_navsat_driver #169: ROS2 Humble, Missing fields in the NEMA message, 'utf-8'

brjthomas, 2023-11-01:
[nmea_serial_driver-1] [WARN] [1698857840.799632234] [nmea_navsat_driver]: Value error, likely due to missing fields in the NMEA message. Error was: 'utf-8' codec can't decode byte 0xd9 in position 5: invalid continuation byte. Please report this issue at github.com/ros-drivers/nmea_navsat_driver, including a bag file with the NMEA sentences that caused it. 

[image]

brjthomas, 2023-11-01:
For reference I am using the GlobalSat BU-353-S4 USB GPS Receiver. https://www.amazon.ca/GlobalSat-BU-353-S4-Receiver-Black-Improved-New/dp/B098L799NH?th=1

Carpediem324, 2023-11-06:
In my case, I used "ros2 run nmea_navsat_driver nmea_serial_driver" not a launch file. So, It worked well

FelixCodeRepo, 2023-11-28:
1. baud rate
2. port name
3. frame_id

closed 2023-12-26 by evenator