ros-drivers/nmea_navsat_driver #171: [bug] Using UDP communication, some statements fail to match because line breaks are not cleanly filtered.

zymouse, 2023-12-14:
## settled
- [x] brach: ros2
- [x] commit: [861323c](https://github.com/ros-drivers/nmea_navsat_driver/commit/861323c83c04ca02454187c85528ddca9a7ef94d)
- [x] Already read the document
- [x] Debugging the code to make sure it's a code bug
- [x] Tested TCP and UDP against each other: no problems with TCP, problems with UDP line breaks `\r
- [x] Grab packets via wireshark:
    - [x] Make sure there are no problems with the data
    - [x] Determine that TCP and UDP are receiving the same data
[image]
## Description of the problem
UDP test: match failed because a statement line break was not filtered cleanly

https://github.com/ros-drivers/nmea_navsat_driver/blob/b168f499349e5f26e3aed9982deb44ed2f41f8e2/src/libnmea_navsat_driver/parser.py#L288

[image]

## Possible causes

https://github.com/ros-drivers/nmea_navsat_driver/blob/861323c83c04ca02454187c85528ddca9a7ef94d/src/libnmea_navsat_driver/nodes/nmea_socket_driver.py#L81

zymouse, 2023-12-14:
@ericperko extremely grateful

ericperko, 2023-12-18:
@zymouse I think you want the current maintainer for this project, @evenator

evenator, 2023-12-26:
@zymouse Your analysis of the situation appears to be correct and I would welcome a pull request to fix the bug.

zymouse, 2024-01-09:
@evenator https://github.com/ros-drivers/nmea_navsat_driver/pull/174
It was a pleasure to maintain the program and make him better.

arbu3k, 2026-10-07:
Probably also fixed by my PR #195 . Found this issue-item after creating the PR.