ros-drivers/nmea_navsat_driver #177: How to improve the topic(/fix) output precision, retaining 10 decimal places

miku54, 2024-05-06:
Hello everyone, please tell me how to improve the topic(/fix) output accuracy to 10 decimal places?. When I run the demo, the altitude read always only has 3 digits after the decimal point. , longitude only has 7 digits of data after the decimal point. How to change to retain 10 decimal places?

OS: [Ubuntu 18.04]
ROS Version [ROS1 Melodic]

evenator, 2024-05-06:
Nothing in the driver itself reduces the precision of the altitude or longitude, so the precision you're seeing is the same precision output by your navsat device.

miku54, 2024-05-06:
> Nothing in the driver itself reduces the precision of the altitude or longitude (aside from the limits of floating point precision in Python itself), so the precision you're seeing is the same precision output by your navsat device.

Now I get it,thank you for your answers.