ros-drivers/nmea_navsat_driver #182: Problem with Numpy

Cavalletta98, 2024-11-04:
Hi, i'm getting this error while launching the navsat driver:
Nov 04 13:57:57 cpr-j100-0538 bash[1444]: [nmea_topic_driver-17]   File "/home/administrator/ros2_ws/install/nmea_navsat_driver/lib/nmea_navsat_driver/nmea_topic_driver", line 33, in <module>
Nov 04 13:57:57 cpr-j100-0538 bash[1444]: [nmea_topic_driver-17]     sys.exit(load_entry_point('nmea-navsat-driver', 'console_scripts', 'nmea_topic_driver')())
Nov 04 13:57:57 cpr-j100-0538 bash[1444]: [nmea_topic_driver-17]   File "/home/administrator/ros2_ws/install/nmea_navsat_driver/lib/nmea_navsat_driver/nmea_topic_driver", line 25, in importlib_load_entry_point
Nov 04 13:57:57 cpr-j100-0538 bash[1444]: [nmea_topic_driver-17]     return next(matches).load()
Nov 04 13:57:57 cpr-j100-0538 bash[1444]: [nmea_topic_driver-17]   File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 171, in load
Nov 04 13:57:57 cpr-j100-0538 bash[1444]: [nmea_topic_driver-17]     module = import_module(match.group('module'))
Nov 04 13:57:57 cpr-j100-0538 bash[1444]: [nmea_topic_driver-17]   File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
Nov 04 13:57:57 cpr-j100-0538 bash[1444]: [nmea_topic_driver-17]     return _bootstrap._gcd_import(name[level:], package, level)
Nov 04 13:57:57 cpr-j100-0538 bash[1444]: [nmea_topic_driver-17]   File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
Nov 04 13:57:57 cpr-j100-0538 bash[1444]: [nmea_topic_driver-17]   File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
Nov 04 13:57:57 cpr-j100-0538 bash[1444]: [nmea_topic_driver-17]   File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
Nov 04 13:57:57 cpr-j100-0538 bash[1444]: [nmea_topic_driver-17]   File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
Nov 04 13:57:57 cpr-j100-0538 bash[1444]: [nmea_topic_driver-17]   File "<frozen importlib._bootstrap_external>", line 883, in exec_module
Nov 04 13:57:57 cpr-j100-0538 bash[1444]: [nmea_topic_driver-17]   File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
Nov 04 13:57:57 cpr-j100-0538 bash[1444]: [nmea_topic_driver-17]   File "/home/administrator/ros2_ws/build/nmea_navsat_driver/src/libnmea_navsat_driver/nodes/nmea_topic_driver.py", line 39, in <module>
Nov 04 13:57:57 cpr-j100-0538 bash[1444]: [nmea_topic_driver-17]     from libnmea_navsat_driver.driver import Ros2NMEADriver
Nov 04 13:57:57 cpr-j100-0538 bash[1444]: [nmea_topic_driver-17]   File "/home/administrator/ros2_ws/build/nmea_navsat_driver/src/libnmea_navsat_driver/driver.py", line 40, in <module>
Nov 04 13:57:57 cpr-j100-0538 bash[1444]: [nmea_topic_driver-17]     from tf_transformations import quaternion_from_euler
Nov 04 13:57:57 cpr-j100-0538 bash[1444]: [nmea_topic_driver-17]   File "/opt/ros/humble/lib/python3.10/site-packages/tf_transformations/__init__.py", line 46, in <module>
Nov 04 13:57:57 cpr-j100-0538 bash[1444]: [nmea_topic_driver-17]     import transforms3d
Nov 04 13:57:57 cpr-j100-0538 bash[1444]: [nmea_topic_driver-17]   File "/usr/lib/python3/dist-packages/transforms3d/__init__.py", line 10, in <module>
Nov 04 13:57:57 cpr-j100-0538 bash[1444]: [nmea_topic_driver-17]     from . import quaternions
Nov 04 13:57:57 cpr-j100-0538 bash[1444]: [nmea_topic_driver-17]   File "/usr/lib/python3/dist-packages/transforms3d/quaternions.py", line 26, in <module>
Nov 04 13:57:57 cpr-j100-0538 bash[1444]: [nmea_topic_driver-17]     _MAX_FLOAT = np.maximum_sctype(np.float)
Nov 04 13:57:57 cpr-j100-0538 bash[1444]: [nmea_topic_driver-17]   File "/home/administrator/.local/lib/python3.10/site-packages/numpy/__init__.py", line 400, in __getattr__
Nov 04 13:57:57 cpr-j100-0538 bash[1444]: [nmea_topic_driver-17]     raise AttributeError(
Nov 04 13:57:57 cpr-j100-0538 bash[1444]: [nmea_topic_driver-17] AttributeError: `np.maximum_sctype` was removed in the NumPy 2.0 release. Use a specific dtype instead. You should avoid relying on any implicit mechanism and select the largest dtype of a kind explicitly in the code.

evenator, 2024-11-09:
Can you provide the following info?

* Operating system and version 
* ROS release 
* How you installed this package (source code or package installer)

Cavalletta98, 2024-11-12:
Ubuntu 22.04
ROS Humble
Package installer

Cavalletta98, 2024-11-18:
i just update the transforms3d to the latest version and fix the problem. Previously i had the 0.3.1 version not compatible with Numpy 2.0

closed 2025-06-01 by evenator